samyk/magspoof #2: Having a license?

ifraixedes, 2015-11-25:
Hi @samyk 

Nice job!

I would appreciate if you can define a license for all of this information; whatever you want, but it would be useful to know which one that leaving it under people and laws interpretations.

Thanks so much to release this information.

madmod, 2015-11-25:
+1 This is an awesome project and I would like to know if I can make my own version of this for friends. Also I'm investigating an iPhone case version powered by the camera flash led.

samyk, 2015-11-27:
I'll start reading up on licenses.

Great idea on one powered by the camera flash LED, that would be very cool. Please share any info if you do create that!

ifraixedes, 2015-11-27:
@samyk thanks for your time to think about it

Luabee, 2016-10-07:
Have you considered a license for this yet? I'm interested in building upon this project and selling it.

samyk, 2016-10-08:
I've created and added a license, and will likely use this for future projects and all previous projects which are not yet licensed.

License added to git and provided at https://github.com/samyk/bpl/

closed 2016-10-08 by samyk

0ki, 2017-09-06:
Is that a joke?