samyk/magspoof #4: Pictures of credit card sized version

madmod, 2015-11-25:
Please consider adding this as it sounds really awesome and I'm sure everyone would love to see it!

0ki, 2015-11-26:
Do you mean the coin? Internet is full of pictures. Also, here's a promo video: https://www.youtube.com/watch?v=gJBZMUuhUrw

madmod, 2015-11-26:
No I already own a coin. It's pretty cool although I haven't gotten the NFC feature to work yet.

There are mentions of parts for a smaller credit card sized version of MagSpoof in the readme and it sounded like @sammyk already built one, however it is not in the video or pictures. I just bought the parts for the larger version showed on the protoboard and plan on making the smaller version on a PCB once I have a prototype working.

samyk, 2015-11-27:
I will put up a schematic and pics when I have a nicer version. Currently the difficult part is keeping the entire thing as thin as a traditional credit card (0.76mm). I'm trying various techniques of PCB etching but simply on thin copper layers, not on FR-4, and very small traces for the micro-sized UDFN and WQFN/WSON packaging.

Currently doing this all at home but open to suggestions/techniques or companies who can run small batches!

lovelaced, 2016-08-17:
Did this ever come to fruition? I'd love to see it.

Joshfindit, 2016-08-18:
In terms of getting it thin enough, how close is the tech that large companies use?
(Apple's lightning-cable video compression board comes to mind)

samyk, 2016-08-18:
Coin is a good example. Their card is just as thin as a normal card and sports BLE, NFC, magstripe and lipo.

loa, 2016-08-19:
Would be nice to get a open source/hardware version rolling. Hate my wallet and that I always have the wrong cards with me.

deltamelta, 2016-09-15:
send me ic's programmed i make you a thin prototypes card size or whatever size u want

samyk, 2016-10-08:
@deltamelta, really? Do you produce thin PCBs?

deltamelta, 2016-10-08:
well yeah contact me at frontbackuser at tuta .io

Graydog, 2016-12-01:
@samyk Maybe use copper wires that are connected in the same alignment as the mag stripes of a regular card, and find a different but small driver that can individually pulse an EM signal to a specific  thread, such that it emulates an actual magstrip whilst keeping the form factor of a credit card?

jLynx, 2018-08-02:
Any updates? This sounds interesting