samyk/magspoof #7: Consider L9110H

shellster, 2016-01-06:
Hi,

I've successfully built a magspoof, but I substituted the L9110H H-Bridge drivr 8 pin DIP package for your L293D H-Bridge driver.  My choice seems to work just fine, and seems to be a much better fit as it is considerably smaller, designed to function on 3.3v inputs, and contains exactly the number of pins needed for the magspoof.  I'm submitting it as a suggestion.  Here's the datasheet: http://www.elecrow.com/download/datasheet-l9110.pdf

samyk, 2016-01-06:
Great! Do you have Eagle? I'd accept a pull request with an alternate schematic using it

shellster, 2016-01-06:
I don't, and I've never learned how to do Eagle CAD.  I may submit a pull when I have time to learn it.

The schematic changes are straight forward for those who are interested:
1) There is no enable pin, so you simply remove the connection from the ATTINY85 on Pin 3 to the H-Bridge Driver.
2) The coil is attached to the H-Bridge on pins 1 and 4
3) Pins 5 and 8 are tied to GND
4) Pins 2 and 3 are tied to the C1 CAP/VCC line.
5) Pin 6 on the H-Bridge is tied to Pin 5 on the ATTINY85
6) Pin 7 on the H-Bridge is tied to Pin 6 on the ATTINY85

sabas1080, 2016-01-27:
#10

daijizai, 2016-05-12:
@sabas1080 thanks for sharing the kicad files. I'm trying to source components to build it, but I'm a complete noob to SMD components, PCB fabrication, and finding the right item. Case in point - I accidentally bought a 0.4 mm ceramic capacitor for the 0.1uF capacitor. Is there a BOM? I've been digging through sites trying to figure out how to determine exactly what I need to no avail. Any advice you can share to help me determine what I need?

sabas1080, 2016-05-13:
@daijizai for the 0.1uF
http://mexico.newark.com/walsin/0805b104m500ct/cap-mlcc-x7r-0-1uf-50v-0805-reel/dp/78Y5293

seach as "0.1uF 0805"

I'm working in the new version with more accessible components

sabas1080, 2016-06-25:
@daijizai I've new pull request with BOM and new version of PCB #11