samyk/magspoof #8: Question about track start sentinels

shellster, 2016-01-06:
Hi,

Sorry if this isn't the appropriate place to ask questions.  As I mentioned in a previous issue, I have successfully created a magspoof.  I used a USB card reader to grab the tracks off of one of my cards.  The Track 1 data begins with the usual '%' sentinel and Track 2 with a ';'.  The track information was added to the magspoof code and uploaded to an Attiny85.  However, when I attempt to use the magspoof on my card reader, everything comes back correctly, except the sentinels for both tracks.  In both cases, they are replaced with a '+' symbol.  Is this intentional?  I'm wondering if I have some sort of timing glitch on the first characters.  I have also, thus far, been unable to read both tracks simultaneously with the card reader, using magspoof.  I am still digesting the code, and it is possible that these are separate issues, but I was wondering if you might have some helpful input.

samyk, 2016-01-06:
I haven't run into anything like that -- in fact, I've never ran into an issue where any of the data was corrupt. The only issue I've ever experienced was the card not being read at all, at which point a second replay typically solves it.

Are you sure the software you're using isn't replacing the characters?

shellster, 2016-01-06:
Pretty sure it isn't the card reader, as it is the same reader that I used to record the tracks originally, which worked fine and correctly.  My magspoof is in prototype stage, so I can probably attribute it to a buggy connection or maybe noise.  It is consistently the first character on the track, and nothing else.  I'm going to play with it some more.  I just wanted to validate that it wasn't by design (I couldn't see how from the code).  It would probably work as the sentinel field isn't necessarily required to be the most common character, but now I know it is on my end.

closed 2016-01-06 by shellster

samyk, 2016-01-06:
Weird. Only thing I'd suggest is playing with the number of leading zeros from 25 to a larger number (like 125), along with the BETWEEN_ZEROS to something larger as well. Would love to hear if you solve it and how, though I haven't heard of this before.

shellster, 2016-01-06:
So after jiggling my coil around, I'm getting occasional correct sentinels.  My coil isn't very big and so I currently have it jammed in the reader.  It has to be exactly positioned.  My current hypothesis is that the signal is very weak and when the card reader detects a "card" it is ramping up detection, which has the effect of it misreading the original character.  Why it is so consistent, I don't know.  Maybe '+' is code for "unable to read character correctly".  I'm going to wrap a nice large coil and properly soldier it to my leads, that will hopefully fix the issue permanently.  I'll report back once I've had a chance to do that.  I may need to consider a separate power source for the coil.  Right now, I'm running it from my teensy 3.1 which is also what I'm using to program the Attiny85.  When the coil is running the LED on my teensy gets very dim, which tells me that even with the cap, it's struggling with the load.

samyk, 2016-01-06:
Are you using a capacitor shown as in the schematic?

samyk, 2016-01-06:
Sorry, I see you mentioned cap -- how big? Try a bigger one :)

kevinaizic, 2018-05-09:
Bit late to the party, but I had this exact issue trying to spoof my student ID which is just track 2. No matter what I tried the first semicolon always became a plus sign or a plus sign would appear in front of the track when I omitted the start sentinel altogether (even when I tried it on track 1 with "%" which made it weirder). Using the MSR90 config tool I figured out "+" is its default prefix for track 3 and by default it doesn't show the start sentinel (but it's "supposed" to be the same as the prefix which is why it doesn't usually matter), so seems like something caused it to think it's reading track 3, somehow.                          
[image]

I actually just got it fixed after hours of fiddling when I found this page, by both arbitrarily increasing the zeros as recommended (not sure if that helped), plus I realized moving the coil around to a different position finally gave a track 1 error suggesting it might have to do with the coil's position since I didn't change anything else, and finally flipping the reader so the coil faced the opposite side bestowed upon me that forlorn semicolon. I don't know how common this issue really is but hopefully there's an answer for the next person.

I'm also just working on a prototype but using a large coil I wound over an iron rod (less than 0.5A  at 3.7V since there are so many turns), plus a jellybean L298N board with pretty beefy built-in caps and high drain LiPo battery, so I feel like it's not  power delivery issue for something so specific and reproducible. My magspoof now works over an inch or so away though, which I'm proud of. Next step is building my own obscenely high current H-bridge and using quite a bit more voltage for the coil to see how much range it can realistically get before qualifying as an EMP... Anyway I can't stress enough that I have no idea what I'm doing, but if I had to guess this issue is likely a flaw in the design of these MSR card readers, or maybe some quirk in the nature of blasting them with radiation from afar instead of their intended use...