samyk/magspoof #17: Powered by CR2032?

aph3rson, 2016-09-16:
It seems as if all of the elements of the circuit (using the alternate motor driver, DRV8835) can run on 3V, and the button-cell has a significantly-higher mAh rating. Is there anything preventing the usage of a coin-cell battery to power this device? Would the field be so small, it'd be practically-unusable?

exploitagency, 2017-04-02:
The RyscCorp v2 version of this uses a CR2032 and a PCB Antenna Coil.
Wireless spoofing seems limited so far.
https://legacysecuritygroup.com/index.php/projects/recent/10-magnetic-stripe/40-rysc-corp-magspoof-v2-first-look

Dehyrf, 2017-07-11:
I put together a low-cost v2 spin of my own, adding a DRV8835, USB for power and on-the-fly reprogramming, and a cr2032 power source: https://circuitmaker.com/Projects/Details/Q-B/MagSpoofv2

Dehyrf, 2017-07-11:
[image]