samyk/magspoof #19: It's sending the same track 3 times in a row

mtagius, 2016-11-22:
I reproduced the circuit on a breadboard and I uploaded the code provided.  I am testing it with a Magtek card reader and when I try to have magspoof send a track of data my card reader says it is sending the data 3 times in a row.  I had the arduino print the data being sent over serial and it seems to only be sending the data once.  Am I making a mistake somewhere?

Thank you

mtagius, 2016-12-01:
I find I am getting different results with different models of Magtek readers.  I believe that was the cause of my problem.

closed 2016-12-01 by mtagius

samyk, 2016-12-01:
Strange. You might want to try changing the timing by increasing or decreasing the delays in the code, as you're essentially emulating a human "swiping" the card.

cheeseandcereal, 2017-04-30:
I've been playing around with a few different card readers, and I believe this issue occurs depending on the implementation of the magstripe reader itself. The difference between magspoof and a physical card is that with a physical card, any particular track can ONLY be read from a single physical location on the read head in the magstripe reader. Magspoof has no physical discretion, and blasts out the magnetic information to whatever is close enough to read it. This means that a single track sent by magspoof can be read multiple times (as different tracks) by a magstripe reader, depending on the implementation of the magstripe reader itself.

There is some great information on magstripe readers here that may help to explain some of this: https://github.com/carlosefr/magstripelib

This is the main problem preventing 100% compatibility between magspoof and every single card reader.

mtagius, 2017-04-30:
That makes a lot of sense.  I think that is the solution.  Would that explain why my USB reader gets the data 3 times but the vending machine I tested this on had no problems and charged my card?  Did it work because the vending machine reader probably can only read one track at a time?  Can magspoof ever work with readers that simultaneously read more than one track?  Thank you

cheeseandcereal, 2017-04-30:
Yes, that would almost certainly be why your USB reader is reading the data 3 times. I know of one magtek reader that alternates the single read head for all 3 tracks, to be able to read all tracks simultaneously. Although I don't have one myself, I imagine the read head is reading each byte sent by magspoof 3 consecutive times, at a different physical location, thinking they are 3 separate tracks since it's reading the same data in 3 different physical locations.

I've found most standard credit card readers will be able to parse magspoof regardless, since they will often accept your card even if it can only correctly read and verify one track. So often, even in worst case scenario, if the card reader is reading a single track from magspoof 3 times, it will simply ignore the 2 invalid tracks and accept the valid one.

I believe most, card readers nowadays have the ability to read multiple tracks simultaneously. The differentiation really comes down to the software implementation and how it chooses to parse/verify/ignore the data read off of the magstripe reader itself.

So while the data read off of a card reader might not look exactly the same as a card swipe which can present multiple tracks at once, the software interpreting and dealing with the data off of the magstripe reader can be written so that it is always correctly interpreted. This is usually the case with regular credit card readers at POS, vending machines, etc, since they generally treat separate tracks as backup redundancy rather than requiring the data to always come in perfectly.

So to answer your question, to get the data to look exactly like a credit card for all readers is simply not possible, although something closer may be possible with multiple coils emulating different tracks simultaneously.

mtagius, 2017-04-30:
Wow, that clears up a lot for me.  I thought I was sending the data wrong.  I really appreciate such a detailed response.  Thank you.

samyk, 2017-04-30:
Do note that I played with the timing and replaying methods quite a bit to find something that worked across the most number of magstripe readers despite having only a single coil. Also note that magspoof works quite a lot better than the multicoil Coin card, likely due to the strong EM force generated. I'm open to changes but you will want to perform significant testing if you do adjust things.

cheeseandcereal, 2017-04-30:
Well like I said, the issue really isn't with the hardware/software implementation of magspoof, but rather the 'issue' is simply the idea of using a powerful EM force to send the data, since that EM force has no physical discretion.

In most cases, the magstripe readers are able to differentiate tracks with the start sentinels, however other implementations of a readers act differently. I've even come across readers that will completey ignore and/or replace the start sentinel read off the card with whatever the reader thinks should be there, based on the physical location where the read head gets the data. With readers like that, there simply is nothing that can be done. 

I can definitely see how a multi-coil solution could work even worse with confounding EM fields. I haven't actually tried that, so my suggestion of multiple coils was more of an untested theoretical.

samyk, 2017-04-30:
I understand, however I'm noting that my implementation was by design to work across the greatest number of PoS credit card readers as possible (and NOT intended for 3-track readers). I found that 2-track readers will often read both tracks from magspoof properly as if the user is swiping their card back and forth, where the first swipe it only reads one track properly and fails on the 2nd, but on the "swipe back" it reads the 2nd track properly while keeping the 1st in memory. I described this in the writeup and is the reason magspoof performs the reverse track playing. 

As I did do significant testing with Coin (two coils in the correct track locations) I found failure rates near 50% on traditional PoS readers, while magspoof works nearly 100% with the reverse-track playing technique.