samyk/magspoof #21: emulate track1 and track2

surajjorwekar, 2017-03-23:
Hello I tried to emulate track1 and track2 at a time. I know you have emulated same but you emulated it in back n forth fashion. I want to emulate it in a single one direction swipe. I did a minor tweak in code but it doesnt work.

`if (track == 0)
  {
   
    // zeros in between
    for (int i = 0; i < BETWEEN_ZERO; i++)
      playBit(0);
  for (int i = 0; tracks[track][i] != '\0'; i++)
  {
    crc = 1;
    tmp = tracks[track][i] - sublen[track];

    for (int j = 0; j < bitlen[track]-1; j++)
    {
      crc ^= tmp & 1;
      lrc ^= (tmp & 1) << j;
      playBit(tmp & 1);
      tmp >>= 1;
    }
    playBit(crc);
  }

  // finish calculating and send last "byte" (LRC)
  tmp = lrc;
  crc = 1;
  for (int j = 0; j < bitlen[track]-1; j++)
  {
    crc ^= tmp & 1;
    playBit(tmp & 1);
    tmp >>= 1;
  }
  playBit(crc);
  }`

samyk, 2017-03-23:
Why is the direction important to you? It's in reverse intentionally so
that it appears as a swipe one way and back, allowing the reader to read
each track upon each emulated swipe. If you swipe one way, both tracks have
to play simultaneously which is not possible with a single coil.

surajjorwekar, 2017-03-23:
>If you swipe one way, both tracks have
to play simultaneously which is not possible with a single coil.

Thank You!!! Got my answer!! Two coils it is.

closed 2017-03-23 by surajjorwekar