samyk/magspoof #22: Plays Track1+2, then ONLY Track 2 every other "press"

cheeseandcereal, 2017-04-30:
With the current implementation of the code, the first time you press the button, it will call playTrack(1), then next time you press the button, it will call playTrack(2) (because of how it keeps track and increments curTrack)

This doesn't make a lot of sense, because when calling playTrack(1), it will actually play BOTH tracks because of this if statement within the method:
`  // if track 1, play 2nd track in reverse (like swiping back?)
  if (track == 0)
  {
    // if track 1, also play track 2 in reverse
    // zeros in between
    for (int i = 0; i < BETWEEN_ZERO; i++)
      playBit(0);

    // send second track in reverse
    reverseTrack(2);
  }`

Then after playing both tracks, next time you press the button it will call playTrack(2) which will ONLY play the 2nd track because that if statement isn't satisfied.

Essentially this means that every other time you press the button, the magspoof will ONLY play track 2. All other times it will play both tracks.

This doesn't make a lot of sense, and it should probably be changed by either:

1. Getting rid of curTrack entirely, and simply replace the line
`playTrack(1 + (curTrack++ % 2));` with `playTrack(1)`

OR

2. Remove the reverse track implementation entirely, then simply call playTrack(1) and playTrack(2) sequentially.

zooyn, 2019-08-15:
Im having the same issue except that my goal is to display both tracks 1+2 consistently after every press of the button. Does anybody know how what configurations I should do with the source code?

I think it has something to do with this part of the code:

void loop()
{

  //for(int i=0;i<10;i++){playTrack(1+(curTrack++%2));delay(3000);}

  sleep();

  noInterrupts();
  while (digitalRead(BUTTON_PIN) == LOW);
  playTrack(1 + (curTrack++ % 2));
  delay(50);
  while (digitalRead(BUTTON_PIN) == LOW);
  playTrack(1 + (curTrack++ % 2));
  delay(400);

  interrupts();
  //playTrack(1 + (curTrack++ % 2));
}

Any help will be much appreciated

Immrspy, 2020-02-03:
Samy has stated that this is WAI in #31 , If you want to just run both at the same time, set the main loop to say "playTrack(1);", then it will do the run 1 forward and 2 backwards scheme.

roboyaar, 2023-01-10:
> Samy has stated that this is WAI in #31 , If you want to just run both at the same time, set the main loop to say "playTrack(1);", then it will do the run 1 forward and 2 backwards scheme.

how should i change the code for cards which have only second line on them?
would you pls help me?