samyk/magspoof #25: issue playing track 2+3 without track 1

herrmanns, 2017-12-18:
hello, i am new to this and would like to understand what i am doing wrong. i just want to use track 2 and 3. the start sentinel of each are ";" 
btw i am using the original magspoof.c of samyk.
// consts get stored in flash as we don't adjust them
const char* tracks[] = {
";672xxxxxxxxxxxxxxxxx=xxxxxxxxxxxxxxxxxx?\0", // Track 2
";01xxxxxxxxxx=xxxxxxxxxxxxxxx=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx==x=xxxxxxxxxxxxxxxx?\0", // Track 3
};

it will only use the 2nd row beginning with ";01 blabla when i press the button. row 1 is ignored.
when i use "%672 blabla it works without any issues. but "%" is the start sentinel of track 1 which i dont need.

how do i have to change code to accept track 2 and track 3 only?

thank you and best reagrds

samyk, 2017-12-18:
MagSpoof is still attempting to play those as track 1 & 2 which won't work as track 1's bits per character is different than track's 2 & 3. You need to make sure tracks[] contains **three** values (first one can be empty), then make sure in the loop change `playTrack(1` to `playTrack(2`

herrmanns, 2017-12-18:
Thank You! I 'm sorry but I have no programming skills :) 
It does the trick. Chaepeau!

closed 2018-12-03 by samyk