samyk/magspoof #27: Read error

blond1, 2018-03-22:
HI, 
I have built my magspoof but it steel doesn't word: every time my readers ( i have 2) say read error
even if the information of the track is correct. I don't understand why. A wrong coil can do this sort of errors?
Thanks for your answer

andyjsmith, 2018-07-18:
Did you ever resolve your issue? I'm having the same problem, getting %E?;E?+E? on my card reader but normal cards work fine as well as using MST on my Galaxy S8.

blond1, 2018-07-18:
Hi
I finally bought a magspoof but i have the same error with mine and the other. Are you under Windows ?do you have telegram or whatapp ?

Télécharger Outlook pour Android<https://aka.ms/ghei36>

andyjsmith, 2018-07-19:
Yeah, I'm on Windows 10. I have telegram.

If I understand you correctly, you have the same error on both your homemade magspoof and the one you bought (Rysc Corp?). Maybe it's an issue then with the card reader itself? I haven't had a chance to test mine out in the wild yet but maybe it works on normal commercial card readers?
Have you tried yours elsewhere?

blond1, 2018-07-19:
Give me your telegram

Télécharger Outlook pour Android<https://aka.ms/ghei36>

samyk, 2018-12-03:
@andyjsmith Do you still have MST? Do you have an oscilloscope or logic analyzer where you could record how it looks? I believe MST came out slightly after MagSpoof and would love to see how it sends data.

blond1, 2018-12-03:
I have an Arduino. Can someone explain me the steps to follow?

Télécharger Outlook pour Android<https://aka.ms/ghei36>

How about an Arduino?

samyk, 2018-12-03:
You could use a coil of wire to connect to the ADC on the Arduino and sample data in. You'd probably want some diodes to protect the ADC pin though as I'm not sure how powerful the signal would be. Have any diodes? Also if you have something like a Teensy that would work even better.

andyjsmith, 2018-12-03:
I'm still having this issue but I can occasionally get it to work, but none of the readers I have tried have picked up multiple tracks, they either pick up the first track or the second, and no other data.

andyjsmith, 2018-12-03:
@samyk Didn't see your earlier reply, but yes I still have MST on my phone, unfortunately I do not have an oscilloscope but I could try a coil connected to an Arduino. You'd have to give me some advice about the diodes though, I have some that came in a kit but don't know which size/rating to use as I'm somewhat new to electronics.

blond1, 2018-12-03:
I will try next week because i will go to Jordan tomorrow.

Télécharger Outlook pour Android<https://aka.ms/ghei36>

roboyaar, 2023-01-10:
how should i change the code for cards those have just second line on them?
would you pls help me?