samyk/magspoof #30: Hotel cards

jarvla, 2018-11-05:
Hi, I tested magspoof works nicely, but I have tried the hotel entrance cards but I do not get any data, probably the reader will not read the data that is not according to the standard. How do I get my hotel card data? Thanks

samyk, 2018-12-03:
What reader and software?

jarvla, 2018-12-03:
> Od: "Samy Kamkar" <notifications@github.com>
> Komu: "samyk/magspoof" <magspoof@noreply.github.com>
>
> What reader and software?

Hello, I've tried Manhattan 460255 and MR300 as a USB keyboard, regular magnetic card reads no data at the hotel. I think hotel systems do not use the ISO standard and if the reader does not see the expected beginning and end, the data will not be displayed.

Jarda