samyk/magspoof #31: Emulation issues using DIY MagSpoof

jLynx, 2018-12-03:
Hi All

So I have just built my first magspoof and seem to be have a few issues.

I brought a magnetic card reader to dump my card details. When I swipe my card with the reader I get:
`%B4548605864141234^JAMES SMITH              ^2107205100000000000000208000000?;4548605864141234=21072051000000000208?`
_(I have altered the data, THIS IS NOT MY ACTUAL CARD DATA)_

Now when I use my MagSpoof I have just created I get this:
Push button once:
`%B4548605864141234^JAMES SMITH              ^2107205100000000000000208000000?;E?+B4548605864141234^JAMES SMITH              ^2107205100000000000000208000000?`
Push button again (2nd time)
`%E?;4548601577143445=21072051000000000208?+4548601577143445=21072051000000000208?`

I would assume I just need to press the button once for it to print out just like it did when I originally swiped my card, but when I press the button the first time, it seems like it just prints Track 1 two times. once I press the button for a 2nd time, it seem to type Track 2 two times.

Does this seem like the correct behaviour, or have I done something wrong here?

Looking forward to getting this working :+1:

samyk, 2018-12-03:
What did you define as track 1 and what as track 2?

jLynx, 2018-12-03:
Hi Samy
So what you are saying makes sense, now just confirming.
So 1 button press runs both tracks "at the same time". Which should work fine.
The 2nd button press is like a backup in case the PoS couldn't read track 2, so it plays it again? 

Then after that 2nd button press, the code restarts and plays the first track again when pressed(3rd press), etc etc etc

Would this be correct?

samyk, 2018-12-03:
Correct.

roboyaar, 2022-12-28:
hi sir
would you please guide me that on what amount did you set the fuse bit during the programming?