samyk/magspoof #34: "Error reading card" on some terminals

jLynx, 2019-02-07:
Hi Samy & GitHub community 

For the most part, I have it working. This seems to work on about 70% of the terminals I have used it on. I have been using it for about 3 weeks now non stop (poor bank account). 
But at some terminals, it just wont work. It says, "Error Reading Card". Do you know what might be causing this problem?
I think could be one of these:

- Coil
- Code
- Motor Driver (DRV8833)
- Power output

I have tested with the original card it has cloned and the card works fine on the specific terminal.
Have you or anyone else run into this issue before and would you have any ideas on what the issue could be and how to fix it?

Looking forward working past this issue so I can leave my credit card in my wallet. 

If anyone has any questions, I would be happy to answer. 

I also have videos of it working and a video of it failing if that would help anyone.

jn234, 2020-10-23:
k

samyk, 2023-07-23:
Could be the terminal not supporting the reversed track, or maybe the terminal requires both tracks to be presented simultaneously, or maybe the combo of two tracks is the issue in of itself. Next time you run into one can you take a pic and get model info as much as you can for it so we can look at it directly?

jn234, 2023-07-23:
thank god your alive

jLynx, 2023-07-23:
> Could be the terminal not supporting the reversed track, or maybe the terminal requires both tracks to be presented simultaneously, or maybe the combo of two tracks is the issue in of itself. Next [...]

Hey Samy! Yep I will do when I have some time to pick up the project again